taHC81/MUST-ESPhome #1: Adding AC Inverter Off / On / Standby possible?

Christyria, 2024-07-07:
Hello,

and thank you for your great Project.
The ESP Software runs here flawlessy on my PV1800 12V 1kW System - but i have an important "Feature Request":

Is it possible (With the correct MODBus Register) to switch the internal AC Inverter On Off and in Stand By Mode? (Stand By Mode= AC is in Power Efficient Mode, it checks every 6 seconds if there is an Load higher than 25 Watts and automatically shuts on the AC Inverting full on, when the Load disappear it goes to Energy Saving)

Because the PV1800 runs on old PB Batterys and i need the switable AC over the ESP Software to turn the AC off, on or in Stand By Mode to save PB Battery Power. The Power Draw in Full on is at about 1,65A and the Power Draw is all 6 Seconds 1A then off.

Many Thanks.

taHC81, 2024-07-08:
Hello, it looks like a register:

20104 |Inverter search mode enable | 0：OFF 1：ON

0: If disabled, no matter connect load is low or high, the on/off status of inverter output will not be effected.                                       

1: If enable, the inverter begins search mode if the AC load connected is pretty low or not detected. The inverter’s “search” mode reduces stand-by energy consumption during no-load conditions.

If so, then please try to add following code into "switch:" section below. It's just a try, I'm not going to test it on my production plant.

```
  - platform: modbus_controller
    modbus_controller_id: must
    id: inverter_search_mode
    name: "Inverter search mode"
    address: 20104
    value_type: U_WORD
```

Christyria, 2024-07-08:
Thank you @taHC81 for the quick reply.

I will test the Register now and i will tell the result.

I tried your Code Snippet and it ESPHome failed here at the value_type: U_Word.
In your Code there is no switch: Section so i added it on the bottom
(And i'm not an programming expert)

But:

With this Code:

  - platform: modbus_controller
    modbus_controller_id: must_inverter
    id: inverter_search_mode
    name: "Inverter search mode"
    address: 20104
    optionsmap:
      "Inverter search mode Off": 0
      "Inverter search mode On": 1

It worked flawlessy! When set to Inverter Search Mode Off the AC imediately turn full on without Search Mode. When i switched off the Load and set it to Inverter Search mode On, Success (With a delay)! Back to Search Mode.

And when i switched a Load under 50 Watts (40 Watt Fan) it still try to spin up but shuts down because of low load, when i set the Inverter Search mode On, the Fan starts fully.

Again Success... :-)

Now please the Modbus ID for the complete AC On Off. There must be an ID to complete turn off the AC in the Inverter to complete save energy, so the Inverter just draw so 150mA or so without AC on.

And with a correct ID i can switch on the AC Inverter and Off from ESP :-)

Christyria, 2024-07-08:
Another Success!

With this Code:

 - platform: modbus_controller
    modbus_controller_id: must_inverter
    address: 20101
    id: inverter_ac_offgrid_work
    name: "Inverter AC Offgrid Work"
    optionsmap: 
      "Inverter AC OffGrid Work Disable": 0
      "Inverter AC OffGrid Work Enable": 1

I can NOW turn OFF and ON the AC Inverting completetely from the ESP!

And it works like a Charm! 

Now the AC is off but the PV1800 is still on and consume very little Amount of Energy!

taHC81, 2024-07-30:
Perfect, should be useful for others as well.