taHC81/MUST-ESPhome #2: Inverter Boot Issue with my RS485-ESP-Module Communication

Christyria, 2024-07-08:
So,
i've found an Inverter Boot Issue with my RS485-ESP-Module Communication to my PV1800 System:

When the Inverter is completetely off and switched on with the ESP Online and RS485 Cable connected and the Inverter will start it 's CPU and is willing to boot up (The three LED are Flashing to the right sideways and then the beep and boot up) it hangs in the Three LED Flashing to the right and cannot completetely boot up as long the RS485-ESP is still connected and try to recieve/send Data to the PV1800.

As soon the RS485 is disconnected the Inverter CPU starts fully and do it' s Beep and start over. Then when the RS485 Cable is connected - all fine, as long the Inverter is completetly offline and booting again with the RS485 connected.

I think the CPU of the PV1800 on my Model will be disturbed when some RS485 communication is send/received when the Unit is in the Boot-Phase.

Is there a "Trick" in the ESP Code so it will detect a PV1800 Boot Up and do a Pause/Halt of his own Communication aslong the PV1800 CPU is fully booted up?

When there is no solution for this i can try to make a Circuit when the PV1800 is switched off so the ESP is switched off and the ESP will switch on in a certain of Seconds and let the PV1800 boot up properly and i think this can help, too...

Thanks :-)

taHC81, 2024-07-30:
Hi, sorry for late answer. I'm running inverter 7x24, so can't reproduce mentioned issue. I'd suggest to power the ESPhome device from the output of inverter, then it will start once inverter is fully online :)

alex-dokienko, 2024-08-21:
@taHC81 
probably not directly related to original issue logged, but similar thing on my side I'm not running all the time (switch on only when I know there will be planned loadshedding) so I picked up issue with charging.
As I could find out once I connect DIY wifi it sounds like not just read only thing but also reset some of my inveverter configuration with some default values. I still need to confirm if all voltage values set correctly for my 5kW inverter with 48V battery, but one particular issue which causing an issue for me is charging source which once device loaded set to OSO (Solar only). Even if I set it to correct value via inverter menu or via HA it doesn't remember selection after restar. For charging power it work fine (remember my selection).

Is there a way to update that setting in yaml or somewhere to set by default solar and utility (I dont' have solart at all so just solar is not an option for me). 

[image]
Would appreciate any guidence.

taHC81, 2024-08-27:
If you set OSO on a inverter itself, does it remember after power cycle?

alex-dokienko, 2024-08-27:
it behaves weirdly, eventually I understood why it was done like that with SNU/OSO sourcing priority to make custom charging setting based on battery voltage, but for me it didn't work for some reason. Change was applied and saved seems like but inverter was ignoring this setting (even if setting was SNU charging wasn't starting up)

So I just commented out all numbers and selects to block any writes and then did factory reset for all settings via  inverter menu (just updating setting via menu on inverter also wasn't helping).