taHC81/MUST-ESPhome #4: Program for ESP8266

alex-dokienko, 2024-08-08:
Hello, 

sorry for probably stupid newbie question.
I'm trying to build similar module like you built, by using uart and wemos d1 mini.
But I'm unclear what program should be flashed on wemos d1? 
Can you please guide me, I'd imagine there should be some code to read from connected pins? or is it handled all automatically with default program?

All I did so far with wemos flashed with micropython and pushed there some basic script to connect to my local wi-fi network, but what else should be done?

Thanks in advance for any guidence.

taHC81, 2024-08-09:
Hello, yes, this is not covered in this repo. Please follow the official docu [how to install ESPhome](https://esphome.io/guides/installing_esphome), then you can download and modify this YAML file as needed and compile + flash to ESP device like below. Please note the first run will download several packages from the internet, so may take longer.

`esphome run esp8266-MUST-PV18.yaml`

Output should looks like this:
[image]

alex-dokienko, 2024-08-09:
Thank you so much for quick feedback.

Managed to update yaml with own configs and add this to my HA, all working nicely.

Once again thank you for effort and sharing this configuration.

closed 2024-08-09 by alex-dokienko