taHC81/MUST-ESPhome #5: The value of the sensor Invertor power

Klassuper, 2024-08-28:
Hello,
I have such a problem with the sensor "invertor power". When running from the grid, in bypass mode, it shows a value of about 65485.
And when from the battery, then somewhere they are correct.
Other sensors work correctly.
Why can this be?
Many Thanks.
[image]

taHC81, 2024-08-28:
Hello, try to add value_type for an Inverter power definition.
```

  - platform: modbus_controller
    address: 25213
    name: "Inverter power"
    register_type: holding
    unit_of_measurement: "W"
    accuracy_decimals: 1
    icon: mdi:flash
    value_type: S_WORD
```

Klassuper, 2024-08-28:
It worked. 
Thank you for your great Project.

taHC81, 2024-08-28:
Perfect, glad to help.