taHC81/MUST-ESPhome #7: accumulated_charger_power sensor does not show data

ctanapik, 2024-09-24:
Hi! I faced an issue with MUST PV18-3224 VPM.
The Accumulated charger power sensor does not provide data of the energy given on the battery

I have this sensor in the ESP config: 
`  - platform: modbus_controller
    address: 15217
    register_type: holding
    name: "Accumulated charger power MWh"
    id: charger_total_mwh
    internal: true
    unit_of_measurement: "MWh"
    accuracy_decimals: 1

  - platform: modbus_controller
    address: 15218
    register_type: holding
    name: "Accumulated charger power kWh"
    id: charger_total_kwh
    internal: true
    unit_of_measurement: "kWh"
    accuracy_decimals: 1
    filters:
      multiply: 0.1

  - platform: template
    name: "Accumulated charger power"
    unit_of_measurement: "kWh"
    device_class: "energy"
    state_class: "total_increasing"
    lambda: |-
     return (id(charger_total_mwh).state * 1000.0 + id(charger_total_kwh).state );
    accuracy_decimals: 1`

[image]

Only once, at the begining, it was set to 0,7 kWh and stays with that value no matter how much I keep charging the battery
At the same time, Accumulated discharger power sensor shows me correct data of discharging the battery

Maybe someone has similar issue and can help me fixing it?

taHC81, 2024-09-25:
Hello, no idea why this should happen. Try to comment out the line "internal: true" for "Accumulated charger power kWh" and check it.

Alex165-cmd, 2024-09-29:
Congratulations. Tell me which sensors are responsible for battery charge and battery discharge? Thank you.

taHC81, 2024-09-29:
> Congratulations. Tell me which sensors are responsible for battery charge and battery discharge? Thank you.

None of them. You have to use some battery meter - Victron smartshunt, PZEM-017...

Alex165-cmd, 2024-09-29:
> > Вітаю. Підкажіть які датчики відповідають за заряд і розряд акумулятора? дякую
> 
> Жоден з них. Ви повинні використовувати якийсь вимірювач батареї - Victron smartshunt, PZEM-017...

And what sensors can be added to the energy panel???

taHC81, 2024-09-29:
Accumulated Charger Power