taHC81/MUST-ESPhome #8: Must  Battery SOC   PH18

1Nikolay, 2025-04-05:
Here's the English translation of the provided Ukrainian text:

Translation to English:

Hello! I want to share some information. I managed to find the SOC (State of Charge) battery register.

I am using: MUST PH18 5448 pro + jk bms JK-PB2A16S20P
The inverter communicates with the battery using the CAN Pylontech protocol.
The battery SOC percentage displayed on the inverter can be found in register 44180 - this is the direct value.

The following configuration was used to add the sensor:

- platform: modbus_controller            # Battery SOC   MUST  PH18  5448 PRO + CAN  battery PYLON tech
    name: "Battery SOC"
    register_type: holding
    address: 44180
    unit_of_measurement: "%"
    accuracy_decimals: 0
    device_class: battery
    value_type: U_WORD

PS:  I think this will apply to inverters in the PH18 and PV18 series.

taHC81, 2025-04-09:
Interesting, how have you found that register? Thanks for sharing this!

1Nikolay, 2025-04-09:
I used a USB-to-RS485 converter and a laptop. A Python script/program was written to scan all registers (from 1 to 65535) via the converter. I set up a data search pattern based on the current battery SOC. Initially, two registers with direct values were found, but one turned out to be incorrect. Further analysis revealed that the block starting at address 44100 contains information related to parameters read from the BMS. Here are the additional registers:

44176 – BMS Voltage,

44178 – BMS Temperature,

44177 – BMS Current (multiplier: 0.1).

PS: If needed, I can provide the script code Python

taHC81, 2025-04-09:
Great job! Thanks for this.

1Nikolay, 2025-11-22:
Hello. 
Here is my script for searching for the desired values ​​in registers. I provide it in two versions.
The search is when you have a constant variable that you want to find on the "entire map" of registers. If your value changes in registers (and in fact too  - very fast) - this is almost impossible. You need to try to look for something that will be stable in the indicators - otherwise there will be too many "coincidences".
I am uses Python 3.13.2  on win10 - for this script

English version: (  i am translate for you .... )
```

from pymodbus.client import ModbusSerialClient
from pymodbus.exceptions import ModbusException
import time

# Connection settings
port = "COM3"  # Your port
baudrate = 19200  # Baud rate
parity = "N"  # Parity (None)
stopbits = 1  # Stop bits
bytesize = 8  # Byte size
timeout = 1  # Timeout
slave_id = 4  # Your RS485 address

# Possible SOC values (now 19%, adjust if needed SEARCH  !!!!!!  )
soc_target = 19          # Direct value (1 = 1%)
soc_scaled_x10 = 190     # Scale x10 (1% = 10)
soc_scaled_x100 = 1900   # Scale x100 (1% = 100)
soc_percent_max = int(65535 * 0.19)  # 19% of maximum 65535

# Search range (tolerance ±5%)
tolerance = 0.05  # 5% tolerance
soc_percent_max_min = int(soc_percent_max * (1 - tolerance))
soc_percent_max_max = int(soc_percent_max * (1 + tolerance))
soc_scaled_x100_min = int(soc_scaled_x100 * (1 - tolerance))
soc_scaled_x100_max = int(soc_scaled_x100 * (1 + tolerance))

# Create client (pymodbus 3.x+ style)
client = ModbusSerialClient(
    method="rtu",
    port=port,
    baudrate=baudrate,
    parity=parity,
    stopbits=stopbits,
    bytesize=bytesize,
    timeout=timeout
)

# Connect to device
print("Connecting to device...")
if not client.connect():
    print("Failed to connect to the device. Check the port and settings.")
    exit()

# Scan registers (1–65535)
start_address = 1
end_address = 65535     
step = 100  # Read 100 registers at a time (maximum for Modbus)

print(f"Scanning registers from {start_address} to {end_address}...")
for address in range(start_address, end_address + 1, step):
    count = min(step, end_address - address + 1)  # Number of registers to read
    try:
        # Read holding registers
        result = client.read_holding_registers(
            address=address,
            count=count,
            unit=slave_id
        )
        if result.isError():
            print(f"Error reading registers {address}–{address+count-1}: {result}")
            continue

[... 104 lines cut ...]
            # Перевірка на пряме значення SOC
            if value == soc_target:
                print(f"Знайдено SOC {soc_target}% (пряме значення) у регістрі {reg_address}: {value} (0x{value:04X})")
            if value == soc_scaled_x10:
                print(f"Знайдено SOC {soc_scaled_x10/10}% (масштаб х10) у регістрі {reg_address}: {value} (0x{value:04X})")
            if soc_scaled_x100_min <= value <= soc_scaled_x100_max:
                print(f"Знайдено SOC {value/100}% (масштаб х100) у регістрі {reg_address}: {value} (0x{value:04X})")

            # Перевірка на відсоток від максимуму (беззнакове значення)
            if soc_percent_max_min <= unsigned_value <= soc_percent_max_max:
                print(f"Знайдено SOC {unsigned_value/65535*100:.2f}% (відсоток від максимуму) у регістрі {reg_address}: {value} (0x{value:04X})")

    except ModbusException as e:
        print(f"Помилка при зчитуванні регістрів {address}–{address+count-1}: {e}")
    except Exception as e:
        print(f"Невідома помилка: {e}")

    # Затримка, щоб не перевантажувати пристрій
    time.sleep(0.1)

# Закриття з’єднання
client.close()
print("Сканування завершено.")

```